uniclogs/uniclogs-hardware #2: Establish OreSat VLAN (aka "Blue Pipe")

glennl, 2017-08-10:
- Essentially the "Blue Pipe" is a 1 Gb/s VLAN between the EB Penthouse rack and EB. This is officially called the OreSat VLAN.

- Also related is the edge router and firewall to get us out to the Internet.

Dave expects to need a pair of SM SFP's to increase the existing bandwidth of the Penthouse switch. Estimated $250 each.

Estimate the bandwidth we need given the devices on the roof only push I/Q samples from the SDR to the server in EB-91, and doing all the signal processing on the server. Expected this will go across UDP.

Describe the protocols and ports we want to move across the VLAN, and what we want to push across the edge router.

We expect only ports 22, 123 (NTP) and 443 across the edge router.

Essentially answer his question "What are you wanting to put on 'my' network?"
NB: Dave expressed keen interest in getting NTP from our time server. Maybe also provide PTP?

Miles started this document; https://docs.google.com/document/d/1XI038USdZ7fAQvtVEo2skS9kDt9Eve8YVRDrkvMqaz0/edit

jameysharp, 2017-08-11:
I assume @heliochronix should be tagged on this issue too, then.

You can't really estimate SDR bandwidth unless you've either already selected radio hardware (in which case you can use its interface speed as an upper bound, like USB2's 480Mbps) or you know what radio bandwidth you want to capture (so you can multiply that by 2 for Nyquist and times, say, 16 bits per sample). Do we have either of those?

I've made some edits to Miles' document, and I'd like to review it again before it's submitted, so please ping me when you're ready?

jameysharp, 2017-08-11:
We should put some thought into how to distribute data between ground stations, since our choice of network protocol there may be of interest to network admins. I guess the draft currently suggests just using HTTP(S).

I'm tempted to suggest peer-to-peer approaches like BitTorrent or the [Inter-Planetary File System](https://ipfs.io/); the latter is a clear winner based solely on its name, obviously. 😉 In both cases, the CAT might see us exchanging some amount of [µTP](https://en.wikipedia.org/wiki/Micro_Transport_Protocol) traffic over UDP, possibly on random ports. For IPFS they might instead see TCP port 4001.

More generally, we need to discuss what our requirements are for telemetry, including threat model (e.g. does any of it need to be secret?). And aren't we also supposed to capture telemetry for other groups in return?

jameysharp, 2017-08-11:
Why do we need a $350 router instead of just using a server with two Ethernet ports and configuring firewall and routing in Linux?

What equipment do you expect to have in the rack downstairs that we'd need a switch for?

glennl, 2017-08-11:
Yes, actually that would be fine. I just threw an SRX100 in there so the firewall / router could be managed completely independently from whatever we put on our LAN. I have been finding used H2 models on eBay for around $100 +/-.

The Linux box is a great solution too, and could also run an IDS like snort. I think we should choose whatever works best for the group supporting our network security.

Regarding the switch: I expect one (1 Gb/s) VLAN to come into our lab space through structured wiring (jack on the wall). On that VLAN we need our server (at 1Gb/s), and the firewall that goes to Dave's LAN, so that adds up to 3 ports, at least two of which are 1 Gb/s. I would also expect there may be other desires over time to put things (like a laptop or two) on our VLAN as well, so I simply would specify an 8 port Gb switch. Those are easily available for $25 on Amazon.

I will post the "Blue Pipe" drawing here in a minute.

glennl, 2018-08-07:
The OreSat VLAN is up and running. On 4-Jul-2018 Miles ran _iperf_ to confirm the connection.
Server to Roof-Rock64: **940 Mb/s**
Roof-Rock64 to Server: **799 Mb/s**
The connection was made using the existing MultiMode fiber connection and SFP's even though Dave mentioned he would change out the connection to Single Mode. I expect we are okay with the bitrate we have, and can just leave it as is.

closed 2018-08-07 by glennl