uniclogs/uniclogs-hardware #7: Stress analysis of EB-ground-station needs to be performed

jalouke, 2017-08-18:
@Joedang you should pick up where I left off. I found out that the wind load of the antenna with ice is 460 lbf. I started a file for a nice approximation and did a static loading of it using the above load. The fixtures at the base may not be entirely accurate (the center post is not fixed or pinned in any way). Please get with @glennl and discuss what needs to be done to get this more accurate.

glennl, 2017-08-19:
### Look at the following options
* Increased pipe diameter of 6061 Al
* Different material other than 6061-T6 Aluminium, such as chromoly [1] steel. It is understood there are consequences for this material
  * Insane cost
  * Tubing dimensions, not pipe
  * Availability lead time (if even possible) in the needed diameter
[1] https://en.wikipedia.org/wiki/41xx_steel

glennl, 2017-08-25:
I have placed photos of the EB weather station taken on 2-Jun-2017 into our OreSat Google Drive under **OreSat EB Ground Station > Weather Station Photos**. These should be useful to understand what is presently on the site.

Joedang, 2017-09-01:
@jalouke were you using the simulation add-in for SW?

jalouke, 2017-09-01:
yup, if you open the part in solidworks 2016 there will be a static tab at the bottom with the analysis that I did.

jalouke, 2017-09-01:
oops, pushed the wrong button

Joedang, 2017-09-07:
I have some questions about the wind loading. @jalouke, did you put the calculations for that 460 lbf figure somewhere? (I'm assuming you found a reasonable source for the drag coefficient for antennas and used the frontal area quoted by the manufacturer.) 

@glennl, I know you showed/told me the manufacturer's quote for the frontal area with ice, but I don't remember where to find that or if I wrote it down at all. Do you have that? Also, do you know the manufacturer/model of the antenna and/or do you have a data sheet for it?

glennl, 2017-09-07:
@Joedang, I burried down here...
https://github.com/oresat/oresat-ground-station/blob/master/eb-ground-station/GS_Structure_Requirements_1_1.md

glennl, 2017-09-09:
@Joedang 
Here are some links I came across pertaining to wind and ice loads:

* [Where we can buy the TIA-222-G standard for big $$$](http://www.tiaonline.org/all-standards/committees/tr-14)
* [An online document describing some stuff in TIA-222-G](http://www.towernx.com/downloads/TIA-222-G_Explained.pdf)
* [City of Portland page on tower structures](https://www.portlandoregon.gov/bds/article/505592)
* [Design or structural evaluation of supports for highway signs, luminaires, and traffic signals](http://highways.transportation.org/documents/sec03-lts-5trev.pdf)
* [Calculating wind load](http://sciencing.com/calculate-wind-loads-wind-speeds-6104140.html)
* [A Comparison of Wind and Ice Loads for the Design of Transmission Lines](http://iwais.compusult.net/html/IWAIS_Proceedings/IWAIS_2011/Papers/P1_34_227.pdf)
* [Ice loads](http://www.ce-ref.com/Load/ICE/ASCE%207_05_ice/ASCE_7_05_ice.html)

glennl, 2018-08-07:
The decision was made to go with a Rohn JRM27510 and loading it with 950 lbm of ballast. This provides Vmax of 97 mph.

closed 2018-08-07 by glennl