uniclogs/uniclogs-hardware #15: Power Board - Rev. 2 wish list

glennl, 2019-03-28:
This is a list of things to be included in revision 2 of the power board.

**_We are declaring that the power board must be moved to Rev. 2 before our satellite is manifested._**

* Soft starts on all P-Channel outputs
* Move the Station Rock-64 UART that communicated with the Tracker Arduino to UART #1.
* Bring UART #2 out on pins for use as a Rock-64 console.
* Anything else?

oliver-rew, 2019-04-19:
We discussed this, but when we first built the board we noticed there was miscommunication between us. You though we were putting in a UART terminal block and I thought you wanted just pin headers. If you want terminal blocks, we should add them.

oliver-rew, 2019-04-22:
We should opto-isolate the GPIO expander from all the power circuits. We could remove the power regulator and pull power and ground from the Rock64, then we can isolate the entire I2C bus and GPIO expander from everything else. The one weird thing could be the ADC, which is on the I2C bus but whose ground it referenced to the power side. Maybe use a i2c isolator?

glennl, 2019-04-24:
> We discussed this, but when we first built the board we noticed there was miscommunication between us. You though we were putting in a UART terminal block and I thought you wanted just pin headers. If you want terminal blocks, we should add them.

The pin header is just fine. Actually I like how you made it fit a common FTDI header, so I would actually now choose this over my original idea. Let's keep it as it is.

closed 2023-01-15 by Benjamin-C